Ylianst/HTCommanderLegacy #1: "Bluetooth LE not found on this computer."

na7q, 2025-01-20:
I get an error that BLE is not supported on my computer. But it does have BLE.

Windows 10.

khusmann, 2025-01-20:
@na7q Do you have bluetooth turned off? Looks like you need to have bluetooth turned on when you launch the app (it doesn't turn it on for you)

na7q, 2025-01-20:
Bluetooth is on. I'm assuming it doesn't like whatever bluetooth board I have.

Ylianst, 2025-01-21:
I will look into this. So far I have only tried on my Windows 11 machine and on my Windows 10 dev machine, I get the same problem. It's my first time with Bluetooth so, I will try to work any issues.

If you get a chance, opt to show the debug tab, switch to it and try to connect. Then, report what you see. Thanks.

[image]

Ylianst, 2025-01-21:
Ok. I got it fixed. v0.2 is now online. Also, make sure you have Bluetooth on here:

[image]

I also noticed that the radio shows up as two Bluetooth devices. I did not experiment with it a lot, I just hit the pairing mode twice and paired with both.

[image]

In any case, hopefully that will do it.

Ylianst, 2025-01-25:
Wrote [documentation on this topic](https://github.com/Ylianst/HTCommander/blob/main/docs/Paring.md). It should help with the pairing process.

closed 2025-01-25 by Ylianst