Ylianst/HTCommanderLegacy #2: Discussion & Updated

Ylianst, 2025-01-22:
Update report. I now mandate that one of the channels be called "APRS" in order to transmit using the APRS feature. If you don't have such a channel, you get notified with a "Setup" button.

[image]

You can edit the channel to fix this, or hit the "Setup" to get it setup for you.

[image]

I try to make it as easy as possible, but feedback appreciated. Now, even with a single VFO setup, the radio will always listen to APRS if setup to do so. I don't change that setting yet, but will probably do so in the future. So, you can listen to a non-APRS channel and still receive APRS messages. When you try to respond, it will send back on the APRS channel.

[image]

I did mess-up during a net tonight and sent APRS message on the wrong frequency. So, I think this change is absolutely required to stop others from doing the same.

Ylianst, 2025-01-23:
Version 0.4 is just out. I added multi-region support, improved Bluetooth setup, improved crash log (debug.log) and "show all channels" menu option.

[image]

[image]

Ylianst, 2025-01-24:
Version 0.5 is just out. Big update of the communication stack to the radio and a bunch of testing. Region support is a lot better now.

Ylianst, 2025-01-25:
Version 0.6 is out. The packet capture feature has been greatly improved. If someone can capture packets of a conversation between a user and a BBS and send the data over, that would be wonderful. I could start work on the BBS terminal support.

[image]

na7q, 2025-01-25:
Please for the sake of APRS standards have it show the RAW unparsed packet as the "data", not the payload. This is one of the reason the HT/BTech app for these radios is so terrible, they removed every standard that is known in APRS and made it worthless.

You've got an amazing start! But please make it show the full packet.

Also, you're just up the road from me, I'm out on the coast. Small world.

Ylianst, 2025-01-25:
Thanks @na7q for keeping me honest. Yes, that is exactly the feedback I want. I will change "AX.25 Payload" to "AX.25 Data". For the full frame, you can select and right click on the upper list of packets and copy full HEX packets to the clipboard, but I will go ahead and add full frames in the bottom windows.

na7q, 2025-01-26:
I LOVE what you're doing and I'm very thankful for it! I'll try to get some more feedback to you when I can. I'm prioritizing my own project within APRSdroid at the moment! So much to do, not enough time!

Ylianst, 2025-01-30:
Version 0.7 is out and we a lot of improvements all over the place. The biggest being the new address book and terminal feature.

You can now save your favorite APRS and BBS stations in the book and when you want to communicate, the software will get you setup automatically.

[image]

Then, we have the new terminal. Supports AX.25 and APRS frames. This should allow you to nicely chat with the APRS BBS, but I will work on adding more protocols.

[image]

Ylianst, 2025-01-30:
@na7q - Oh, I also made fixes to the packet capture feature :) Hopefully it's to the level now.

[image]

na7q, 2025-01-31:
I will hopefully have some time this weekend to play with it. Thanks!

Ylianst, 2025-02-05:
Version 0.8 is out. You can now import, export and drag & drop channels to reprogram the radio. I also added a built-in web server, but that does not do much right now, just display debug messages.

[image]

na7q, 2025-02-05:
Great! Haven't dived into this yet. But looking forward to it. I've had a few surprise days off from work so I've been playing catch up with APRSdroid updates.

Thanks for the work you're putting in!

Ylianst, 2025-02-06:
I was in Las Vegas over the weekend and got this APRS map. Pretty good range.

[image]

na7q, 2025-02-06:
Some suggestions with the "data". 
As displayed..
[image]

As it should be displayed.
NICOLI>APRS,WIDE2-1,KNAPPA:T#081,167,095,011,065,128,00000000

Whenever an ssid is 0, we remove it from the readable text. We also take the source (NICOLI) and place that first, followed by a > then the the destination (APRS), followed by the rest of the path or paths (WIDE2-1) separated by a comma for each one. You should also remove the spaces from the way you have it. One the paths have ended, you then add a colon.

The seventh (6) byte in the digi paths need to have a * when they have been digipeated. So for example in my python programs I do it like this. Though it's probably not the best way. Amateur coder here. Hopefully this makes sense. Other than that, this is looking good.

                                # 7th byte carries SSID or digi:
                                seven_chunk = path_chunk[6] & 0xFF
                                                        
                                if seven_chunk & 0x80:
                                    digi = True

                                #print("Digipeated: ", digi, "\n")

                                # Check if digi is True, then append '*' to the path_address_with_asterisk
                                if digi:
                                    path_address_with_asterisk = f"{path_address}*"
                                else:
                                    path_address_with_asterisk = path_address

na7q, 2025-02-06:
I'll also get back to you on some suggestions with messages and other stuff. We appreciate what you're doing!

Ylianst, 2025-02-08:
@na7q - Version 0.9 is out and I added your suggestions for displaying APRS packets in the packet capture tab.

[image]

In addition, I added a built-in BBS which is right now, just an text adventure game. The BBS only supports raw AX.25 frames for now, I will add APRS in the next rev. You can activate and see user stats on the BBS panel.

[image]

The terminal was also improved a lot, colors and packet handling.

[image]

Ylianst, 2025-02-13:
Version 0.10 is out. I added a APRS stack so now, there is retry and ack support for APRS messages. I also added the ability to put HT Commander on the tool tray and if you get a APRS message directed to you, it will notify you on the desktop. I also added a bunch more documentation. Also, [The Comms Channel made a video on HT Commander](https://www.youtube.com/watch?v=McIHckn4Kr4&t=204s). So, lots going on.

autocharg, 2025-02-28:
Hello All
KC5OLH
new to this github thing and back in the day i studied VB6, but decided programing was not a carier choice for me. i posted in the wrong section the other night i gues,I was excited and had to get the news out. here is the cut and past. hope it helps.
I may have found something.
 i tried for hours to get the UV-pro to reconnect after once having it connected and running.
 what worked was UV-pro device was connected"paired all ready.
 i got my cell phone "android" and connected the HT app made sure it was running.
 windows 10 setup add bluetooth device is ready click add device , Bluetooth :
 show all devices was displayed. disconnected HT app on cell phone and UV-pro pop'ed up
 immediately click connect. that was it your commander app started working.

ps: playing with the BBS setup are their any instructions i have over looked?
kc5olh
73

Ylianst, 2025-03-01:
I have limited time these days, but did a little bit of work on getting the Winlink user interface built into HT Commander. Here is what is looks like. Keeping it nice a simple, but the code behind the UI is quite significant. I got the Winlink password hash, CRC and compression all working. I still have to put a lot of pieces together.

[image]

Ylianst, 2025-03-01:
@autocharg Thanks for the write up! It does seem like it's difficult to pair this radio with Windows. I am going to have to take another looks at Bluetooth pairing. All this information helps a lot.

For the BBS, right now you pick a channel on VFO1 the main channel and then go to the BBS tab and hit "Activate". That is it. Any incoming packets or connections will be accepted. Right now, I have nothing but the text based adventure game. I personalty have two radios, so I use one HT Commander in terminal mode to connect to the other HT Commander is BBS mode. I will add more standard BBS features in upcoming versions. Note that the BBS will accept RAW X25, APRS and X25 Sessions packets and response in the same packet format.

autocharg, 2025-03-01:
Ylianst
I'm looking at the BBS as an Em-comm tool. Were bulletins / short messages can be posted and revived for some period of time (7 to 14 days). and maybe threads can be created and stored. Also this can be done on VHF/UHF with Btech but what about HF suport that would require some Rig control and greatly expand the scope of this project. VHF/UHF is good enough for local coms. The biggest challenge I'm facing is ease of use, right now finding a cell phone app that can connect to a VHF/UHF radio via usb-c or better yet Bluetooth with a terminal screen is changeling. so far aprsdroid looks like an option but its bloated and not user friendly and sends out your position all the time. My goal would be to have the average person to be able to turn radio on and start app. thats it. offcorse their would be some setup during installation. If it's not simple most people wont do it. (KISS)
I think what you got going on so far is great keep it up.
KC5OLH
73

Ylianst, 2025-03-01:
@autocharg I am completely in agreement with you. Ease of use is a big issue. I would like users to focus on the communication and have the tool and hardware be as simple and integrated as possible.

For bulletins / short messages, my view is that this is the sort of thing you should compose offline and sync with the BBS. The terminal experience is not a good one. A digital sync can easily figure out what is new since last time you did a sync and be very efficient with compression and frequency usage. You can then read the bulletins offline.

I am thinking to integrate bulletins and short messages within my mail user interface, but would be nice to have a common format.

autocharg, 2025-03-01:
I agree , I haven’t see any android apps the can do this.
If you need Bata testers, I’m here. I used to be a field engineer for the DOD. UAV’s and such, the communications and net working/pc maintenance side.
KC5OLH
73

codywsmb02, 2025-03-06:
Hello!

I am trying to import channel files from RadioReference.com into the HT commander. However, the popup screen that should show up to drag and drop channels isn't showing. 

I have tried deleting the program, and redownloading it. I have also tried exporting the channels that are on the radio, then re-importing them, but it didn't pop up the window to drag and drop the channels either.

I included both files that I am trying to use. Please let me know if I can provide anything else to assist. 

Thank you so much!

Cody

[channels.csv](https://github.com/user-attachments/files/19100166/channels.csv)

[FLFrequencies.csv](https://github.com/user-attachments/files/19100167/FLFrequencies.csv)

Ylianst, 2025-03-06:
Oh nice. Thanks for sending the files. I will look into this. I am working on Winlink support and it's starting to be crazy nice, but yes, channel programming is also super important and will work on it.

Ylianst, 2025-03-06:
@codywsmb02 - I just tried it out. channels.csv is a format I should support and it worked for me.

[image]

The other file, FLFrequencies.csv is not a format I know anything about, but I will add support for it. Can you confirm channels.csv does not work for you at all? Nothing shows up? - Thanks!

codywsmb02, 2025-03-06:
Thank you for the quick response!

I have tried both CSV files, and after the import dialog window closes the popup window doesn't appear to show the channels that I am trying to import. It's as if I never selected to import anything. 

I tried deleting the zip/extracted files, rebooting and redownloading and launching the HT application, but the same issue occurs. 

I also tried factory resetting the radio, but same issue. I am using Windows 10 latest updates, and 0.8.0-1 firmware on the UV-Pro. 

Thank you!

Cody

Ylianst, 2025-03-15:
Ok, it took a long time, but 0.14 is out with Winlink support. A added support for the client to send/receive emails along with Winlink gateway support in the BBS. Now, I don't support connections to the Internet yet, so, if the BBS gets Winlink emails, they go in the inbox or outbox depending on who the email is for. So, you could setup the BBS somewhere and have people check-in and send each other emails and the BBS would act as a private email router.

[image]

Ok, now I can work on more channel setup.

Ylianst, 2025-03-16:
I just found this [video in French on HT Commander](https://www.infothema.fr/forum/index.php?topic=5180.0). Nicely done. This way before the mail feature was added.

autocharg, 2025-03-16:
Hello
I just discovered that WOAD has a Terminal screen/ session in it. I’v been using it to access my BBS the last two days. Currently I’m running BPQ32. I’m having trouble getting people to use the bbs right now how about you. Do you have any testers? I’m thinking of putting together some training material for the AKA bofang users of the club. My local club dosen’t seam to support the “ digital modes” too much. Maybe its a lack of knowledge . Got to KISS this i think.
KC5OLH, 73

Ylianst, 2025-03-16:
@autocharg - Funny you mention the terminal in WOAD, I saw [the video from The Tech Prepper](https://www.youtube.com/watch?v=pcdh4R6kdvo) just last night also discovered this terminal. As a result, I was able to test my own terminal and just uploaded HT Commander v0.15 with terminal improvements. I can use WOAD terminal to talk to the HT Commander terminal or BBS and it now works well.

So yes, I am certainly hoping that HT Commander be part of the tools that will enable people to have fun with digital modes. I don't have many testers, I feel like I just got started with this project. Having a presentation about digital modes would be great, if you have something you started on already, let me know, I will gladly contribute to it. Having people be able to play with APRS, Terminal, BBS and Winlink would be great. Also, if you have any ideas to make this even more fun, I am listening.

Ylianst, 2025-03-16:
FYI. HT Commander v0.15 was released with terminal improvements. It should be able to connect to BBS'es pretty well. Feedback appreciated.

autocharg, 2025-03-16:
Hello
The attached pdf is a High level presentation to get people interested . I will be using it at my next clubs meeting. whish me luck. lol
Feel free to change in any way.
KC5OLH 73

Ylianst, 2025-03-17:
@autocharg - Sadly your PDF did not attach to the Github response, so, I can't see it. Please do post on Github or mail it to me so I can work on posting it if possible. Thanks.

autocharg, 2025-03-17:
Sorry about that, here it is.

Ylianst, 2025-03-18:
@autocharg - Thanks. I got your PDF. Is it ok if I post it here?

I just released v0.16. I tested Winlink with my local official RMS server and ran into one issue that I fixed. So now, I can send emails and it works great.

autocharg, 2025-03-18:
Sure go ahead and post it, if you need to change anything do that also.
Like I said it a high level understanding of things.

Ylianst, 2025-03-23:
For fun, I just added a Bittorrent like file exchange protocol to HT Commander. Pretty simple to use, but work in progress especially timing of packet transmits.

[image]

autocharg, 2025-03-23:
Hello
Are you doing this via Full Duplex, or just simplex. I think a file shearing program that could take advantage of Full duplex would be game changing.

Ylianst, 2025-03-23:
@autocharg - The radio is half-duplex. Unless I am missing something, I can't transmit and receive at the same time. For repeaters, you can transmit on one frequency and listen on another, but it's still half-duplex.

The main benefit here is taking advantage of the built-in multicast of radio. You can send a data block once and have 10 stations receiving it, so, it's efficient in that way.

autocharg, 2025-03-24:
Just up dated to .17 program installed and launched looks good. Could not connect to radio.  Went back to  .13 and was able to connect to radio. Went back to .17 and was unable to connect. Also when i went to File, settings and enabled radio to transmit after clicking ok. the program flickered and the connect button at the bottom of the radio image was gone. Reversed the enable radio to transmit and the button remained gone. returned to .13 and all is good.

73

Ylianst, 2025-03-24:
@autocharg - I just noticed that my 0.17 MSI installer had missing files and did not work. I just fixed it, let me know if it works. I also code signed HTCommander the executable and MSI installer. So yes, please try again and let me know. 0.17 has a lot of improvements, I don't think 0.13 has Winlink support.

autocharg, 2025-03-25:
Hello, I had time to do some testing tonight.
Had troubles, When selecting a station using the "Mail" screen I found that it dose not automatically change the channel on the radio. plus my outbox still shows 1 msg. after a connect. below is the debug info. I used winlink and varaFM to verify radios were making connections.
Looking for compatible radios...
Found UV-PRO - 38D20000F01A
Attempting to connect to radio MAC: 38D20000F01A
Selected device: UV-PRO - 38D20000F01A
Connecting to radio...
Getting radio service...
Getting radio characteristic...
Getting radio information...
X25: Connect
X25: ClearTimer Connect
X25: ClearTimer T1
X25: ClearTimer T2
X25: ClearTimer T3
X25: ConnectEx
AX25 CONNECTING
X25: ClearTimer Disconnect
X25: ClearTimer T3
X25: EmitPacket
X25: Renumber
X25: ClearTimer Connect
X25: SetTimer Connect to 10000ms
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=15
X25: Timer - Connect
X25: ConnectEx
X25: ClearTimer Disconnect
X25: ClearTimer T3
X25: EmitPacket
X25: Renumber
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=15
X25: Timer - Connect
X25: ConnectEx
X25: ClearTimer Disconnect
X25: ClearTimer T3
X25: EmitPacket
X25: Renumber
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=15
X25: Timer - Connect
X25: ClearTimer Connect
AX25 DISCONNECTED
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=15
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=15
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=15
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=15
X25: Connect
X25: ClearTimer Connect
X25: ClearTimer T1
X25: ClearTimer T2
X25: ClearTimer T3
X25: ConnectEx
AX25 CONNECTING
X25: ClearTimer Disconnect
X25: ClearTimer T3
X25: EmitPacket
X25: Renumber
X25: ClearTimer Connect
X25: SetTimer Connect to 10000ms
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=15
DataFragment, FragId=0, IsFinal=False, ChannelId=27, DataLen=53
DataFragment, FragId=1, IsFinal=True, ChannelId=27, DataLen=16
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=37
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=30
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=32
X25: Timer - Connect
X25: ConnectEx
X25: ClearTimer Disconnect
X25: ClearTimer T3
X25: EmitPacket
X25: Renumber
[... 44 lines cut ...]
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=15
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=15
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=15
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=15
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=15
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=15
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=15
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=15
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=15
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=15
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=15
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=15
DataFragment, FragId=0, IsFinal=True, ChannelId=27, DataLen=15

from winlink
*** Receiving HDD83R8WFQL3
*** HDD83R8WFQL3 - 412/323 bytes received
*** Bytes: 375,  Time: 00:06,  bytes/minute: 3625
   FF
FQ
*** --- End of session with AB5MC-10 at 2025/03/25 00:33:19 ---
*** Messages sent: 0.  Total bytes sent: 0,  Time: 00:46,  bytes/minute: 0
*** Messages Received: 1.  Total bytes received: 375,  Total session time: 00:46,  bytes/minute: 482
*** Disconnected from Winlink RMS: AB5MC-10 @ 2025/03/25 00:33:21
*** Session: 0.8 min;  Avg Throughput: 51 Bytes/min;   1 Min Peak Throughput: 51 Bytes/min

Ylianst, 2025-03-25:
@autocharg - In the mail tab, can you select "Show Traffic" and then hit connect? That should show what is going on.

[image]

As for not changing frequency, that is odd. Are you using multiple regions? I have not worked on this much yet.

Ylianst, 2025-04-03:
[HTCommander 0.19 is out](https://github.com/Ylianst/HTCommander), you can listen to audio but not transmit yet. I mute the audio if it's coming from a channel that is muted, otherwise, works as expected. The UI is very basic right now, just enable it in the Settings menu.

Audio support is a big step since it enables a bunch of new possibilities.

Ylianst, 2025-04-06:
[HTCommander 0.20](https://github.com/Ylianst/HTCommander/blob/main/README.md) is out with DeepSpeech integration. It will transcribe audio from the radio into text in near real time. You need to download to additional large files to make this feature work and there are imitations, but it's fun to see in action. [The details are here](https://github.com/Ylianst/HTCommander/blob/main/docs/Voice.md), feedback appreciated.

[image]

RodMyers, 2025-04-12:
Seems the links to the zip andmis installer are not linked to anythins

Ylianst, 2025-04-12:
@RodMyers I just released [HT Commander v0.21](https://github.com/Ylianst/HTCommander) and I fixed the links.

This version have Open AI Whisper added in and it's quite impressive. It will do speech-to-text and text-to-speech in the voice tab. You can sort of transmit using the computer's microphone, but good support for that will be in a future version.

[image]

mwconstruction, 2025-04-13:
First, thank you for your work on this project.

Trying today on version 0.21 (same result with prior version)
Connected to both bluetooth instances of the VR-N76.  No response when clicking connect in HT Commander. Any suggestions?

[image]

Ylianst, 2025-04-13:
@mwconstruction Looks like you paired the device correctly, so, it's got to be something wrong with HT Commander.

Do you know what firmware you have in the radio? I did test on v0.8.0 and above.

Next, in the debug tab, can you select "Show Bluetooth Frames", then, try to connect and send over a screenshot of what you see?

[image]

I still have a few essential features to work on around audio control/routing/transmit, etc. Once I get that done, I will start asking for local ham's in my area to help test and debug. I have two UV-Pro's myself, I don't have a VR-N76, but I will see if anyone has one around for a debug session.

mwconstruction, 2025-04-13:
Thanks for the quick reply!
Firmware 0.8.3-1

I am now able to connect after a couple of changes:
The developer debug showed no activity in the log at all when clicking connect. After a reboot of the PC, another attempt to connect HT Commander, the log showed activity - failed to connect message (progress).
Realized my phone may be using the VR-N76 Bluetooth connection already, preventing the PC from connecting. 
I turned off Bluetooth on the phone. 

Success! Connection established. I'll play around with the app tonight and see how things work for me. 

[image]

Ylianst, 2025-04-13:
Interesting. In the last two versions of the app, I display this dialogbox when connection failed:

[image]

I will review the code to see there are more situations I don't handle where this message should be showing up.

mwconstruction, 2025-04-13:
I am getting that dialog box now when I intentionally connect the radio to my phone first.

Ylianst, 2025-04-13:
@mwconstruction - Good, at lease some things work. I may release another update to the speech-to-text this evening.

mwconstruction, 2025-04-14:
Success importing channels from a Chirp CSV file (built off Yaesu FT-4x codeplug) into HT Commander. Overall things went very well, thanks for the great app!  

The only undesired behavior I noticed so far was on channels with different rx/tx tones. It failed to pick up any tones for the imported channel, and removed the offset for transmit frequency. 

If there is a way to name regions (channel groups) from HT Commander, I missed it. I named my groups using the manufacturer's app.

Ylianst, 2025-04-14:
I am new to Ham radio and probably messed this up. If you cut & paste a CSV file along with that the result should look like, I will fix it. Also, I have to work on region support. I keep using a single region on my radios, but it needs to be worked on. Thanks for the feedback.

mwconstruction, 2025-04-15:
[image]

[image]

[image]

[image]

[image]

My screen grabs are the channels as they should be imported. See CSV file below if you want to test how they come out after import to HT Commander.

Channels 1 and 2 are examples of Cross Tone using DTCS Codes. Notice that Chirp added matching rToneFreq and cToneFreq values of 88.5 when it exported this CSV file. Those are spurious and should be ignored. 

Channels 3 and 4 are examples of Cross Tone using standard PL aka CTCSS tones. Notice that Chirp added matching Dtcs Code and RxDtcsCode values of 23 when it exported this CSV file. Those are spurious and should be ignored. 
A Tone column value of "Cross" would suggest that the spurious addition of matching tones to either the CTCSS fields or DTCS fields are bad data. Whichever has different values for transmit and receive tones is the data to use. 

I hope I explained that clearly, please reach out if I need to do better there.

I noticed an additional issue while writing this up. 
Look at channels 3 and 4. They are split frequency repeaters that do not use the typical "Offset" value. Instead, the frequencies are an arbitrary distance apart. When importing these two channels, HT Commander adds them as simplex instead of duplex, failing to grab the transmit frequency from the "Offset" field.

Thanks again for your excellent work. Let me know if I can be of any assistance.

[2025.3.28 Personal FT4X.csv](https://github.com/user-attachments/files/19746433/2025.3.28.Personal.FT4X.csv)

Ylianst, 2025-04-15:
Thanks you so much @mwconstruction. One quick thing I notices is that #3 is NFM, or Narrow FM, so I imagine that should be reflected in the radio. Thanks for mentioning the spurious data, that helps a lot.

I got speech-to-text improved a lot in the upcoming version, I had it running all day. So, that and your fix will be a good improvement. I could have a bit of time tomorrow to work on this.

Ylianst, 2025-04-16:
@mwconstruction - I just released HTCommander v0.22 with the channel import fix. Let me know if it works for you. Also, big thank you for your write up. I bet a lot of people had issues but don't report it, thanks for much for doing so in such detail. Let me know if you see anything else I could fix.

mwconstruction, 2025-04-17:
Gave it a test drive, so far so good. I'll let you know if I run into other issues. If you wanted to take on incorporating the radio settings, the programming software from the manufacturer has lots of room for improvement on layout and useability. That would  be a pretty big undertaking to get right though.
I figure if you're writing free software for the community, the least I can do is give you clear information to work with. Thanks for the fix

codywsmb02, 2025-04-17:
Hello!

First of all, this is AMAZING work, and you being able to add the CHURP.csv channel files is so helpful!

I am having issues with adding AM channels for the FAA/Airport Frequencies. 

The range in the dialog box is 136-174 MHz and 300-550 MHz, but the Airport Frequencies are around 118 MHz.

There's no option to enter it in from HT Commander, but I can manually enter it into the radio itself. I'm using the UV Pro. 

I attached a picture of the channel I cannot add. 

[Image](https://github.com/user-attachments/assets/afbc8fef-0a85-4e9a-92cd-6fdc80652973)

Thank you!

Cody

[22 more replies]