Ylianst/HTCommanderLegacy #3: add support for Linux

sysadminpower2019, 2025-02-13:
I saw your comments that you would be willing to add support for Bluetooth on linux. 
I think the community would really appreciate being able to run your fantastic app also on linux.
let me know if i can help in any way.

thanks

Ylianst, 2025-02-13:
Thanks. I will look into is at some point. I think a lot of the ham radio community is on Linux, there is a weekly Linux net in Oregon where I live so, not a surprise that would be a request.

DanDawson, 2025-02-13:
Thank you for building this! I've very aware of how much work porting applications to other platforms is, but I will agree that many hams are now using Linux, such as Raspberry Pi for example, and also using Mac computers a lot more than in the past. 

So if you are going to undertake the effort, I would definitely suggest you consider building in a platform that could be used across a variety of platforms. One option might be to follow the route of having the actual software run behind the scenes, and have the interface for it be in a web browser, like [Pat Winlink](https://getpat.io/) (built with Go) or [CalTopo Desktop](https://training.caltopo.com/all_users/desktop) (built in Java) for example. 

Some options to consider on the back-end would be Go, JavaScript (Node.js) or Python for example, and they would run a small local web server and your interface would be html/css built with any modern framework such as ([Bootstrap](https://getbootstrap.com/), [Tailwind CSS](https://tailwindcss.com/), [Bulma](https://bulma.io/), [UIkit](https://getuikit.com/), [Material UI](https://mui.com/material-ui/), or [Foundation](https://get.foundation/).

khusmann, 2025-02-13:
@DanDawson I've put together a cross-platform python package called [benlink](https://github.com/khusmann/benlink) that gives you full control of the radio and would be perfect for the exact architecture you describe. It's just waiting for someone with a vision like yours to build a UI around it ;-)

It runs on Linux / Raspberry Pi, Windows, OSX, and even python-for-android (in theory). It gives you an async API for controlling the radio you could easily build a frontend on in any modern web framework.

Ylianst, 2025-02-16:
@sysadminpower2019 @DanDawson. After doing a bunch of changes, I did get HT Commander compiling on Linux. I had to make some changes to the UI and I probably need to do more as, in my view, the UI does not look as good yet. Bluetooth code is OS specific so, I have to re-code that part, but at least the UI is working ok.

[image]

[image]

DanDawson, 2025-02-16:
I just wanted to take a moment and thank @Ylianst and @khusmann for their work on [HTCommander](https://github.com/Ylianst/HTCommander) and [benlink](https://github.com/khusmann/benlink). These radios are amazing, but what is truly most exciting about them is the ability for the community and developers to come together and expand the capabilities even further. Nice work!

elXaptation, 2025-08-03:
Ant chance HT Commander works using Wine on linux? Gonna try when I get some time..

IstiCusi, 2026-01-02:
Dear Yilanst, is there a platform independent (aka linux-compatible) version or a wine installation guide (would be at least a workaround) already. It is a great application, but we miss really the linux version for outdoor applications etc. Cheers and thanks a lot for all the effort .. maybe we can help establishing such a port. But we would need advice.

Ylianst, 2026-03-24:
So yes, I sort of want to do a MacOS and Linux port and I looked at Avalonia, but it does not have the MacOS style, I also was hesitant because of the Avalonia license so, opted to try something else first. Flutter may be better, I have not looked yet. Another route I may take is the Progressive Web App (WPA), it can't do audio but will work on all platforms and I am thinking this is plenty good for doing authenticated AFSK messages back to my home assistant. If you have ideas, let me know.