Ylianst/HTCommanderLegacy #4: VR-N76 - No Incoming Messages

mkturner89, 2025-02-13:
I picked up this software and paired it on my windows PC with my VGC VR-N76.
All features seem to be working, BUT when a packet is received that is destined for me, I do not see it in the APRS section.
It is seen in the packet capture section, but nothing in the actual APRS section.
I'm using WXBOT to test.

[image]

[image]

I'm running version 0.1 of the software, and the VGC has the newest firmware.
The APRS callsign in the HT setting and the software match.

Ylianst, 2025-02-13:
@mkturner89 Oh! This is interesting. Also, looking at the WXBOT response, I wonder if I should not merge the messages into a single blue bubble. In any case, in the packet capture tab, can you select the range of packets (your requests and the responses) then right click and save the packets into a .csv file and send it over? I will replay the packet in my dev environment and fix this. Thank you!

mkturner89, 2025-02-13:
Of course! Here you go!
Though my only option was save as .ptcap and github is not letting me attach that file type.
Here's a dropbox link to the file: https://www.dropbox.com/scl/fi/3k2koh9xhsvu0h4cmbdhd/APRS_VR-N76.ptcap?rlkey=8n1ralbeefgkk7e1oamboc55o&st=jtlqp2rl&dl=0

Ylianst, 2025-02-13:
@mkturner89 Thank you. I see this is a 3rd party relayed message. Matches this below. I will fix it later today. Thanks for sending the capture file, it makes work super easy.

[image]

mkturner89, 2025-02-13:
This is an awesome program and I am HAPPY to help.

Ylianst, 2025-02-14:
Ok, I got it fixed and at the same time, discovered WXBOT. Very nice!

[image]

I think I am going to have to remove duplicate messages even if coming from different stations. I will look into this.

mkturner89, 2025-02-14:
It works now! Thanks!
There's other bots out there too, but WXBOT is the one I use most often
If you need anything else tested on the N76, feel free to drop me an email. mkturner89@gmail.com

closed 2025-02-14 by mkturner89

Ylianst, 2025-02-17:
@mkturner89 FYI. In the upper right menu in the APRS tab, there is new a "Weather Report..." option with a dialog like this:

[image]

If you can give it a run and let me know if it works for you, that would be great. Thanks.

mkturner89, 2025-02-17:
This seems to be working well. Tried a bunch of different request combinations and they all worked.

Ylianst, 2025-02-17:
Sweet! Thank you!!!

Ylianst, 2025-02-17:
I just added instructions [https://github.com/Ylianst/HTCommander/blob/main/docs/APRS-Weather.md](https://github.com/Ylianst/HTCommander/blob/main/docs/APRS-Weather.md).