Ylianst/HTCommanderLegacy #5: APRS Beacon Settings

mkturner89, 2025-02-14:
It would be nice to have this program handle the APRS beaconing the same way APRSDroid or the HT app does.
When using just the radio, it will beacon out regularly according to the settings on the device.
When the HT app or a third party program takes control, then the beacon settings are overridden with what the program settings are.
Being able to set the beacon interval, symbol, voice frequency, and comment field would be great.

Ylianst, 2025-02-17:
@mkturner89 - Thanks for the request. I am looking into this. I think the radio's beacon settings continue when HT Commander is connected, however, I can read and change these settings. I released v0.11 where in the "About" menu, there is "BSS Settings..." and you can see all the beacon settings. I will work on offering a way to change them in an upcoming version.

[image]

mkturner89, 2025-02-17:
Cool. This would certainly be nice to be able to change, since I usually like different beacon settings when I'm sitting at my PC, versus when I have it portable or mobile.

Ylianst, 2025-02-17:
Yes, once I get more features in the BBS, being able to beacon the BBS using APRS when enabled is one of many good usage.

Ylianst, 2025-02-21:
@mkturner89 - Version 0.12 is out and I added beacon settings. Let me know if this works.

[image]

[image]

mkturner89, 2025-02-21:
At first I thought it wasn't working, but then I realized it is, but it's not showing up as an outgoing message in the packet log.

[Beacon_Test.txt](https://github.com/user-attachments/files/18911937/Beacon_Test.txt)

[Beacon_Test.ptcap.txt](https://github.com/user-attachments/files/18911952/Beacon_Test.ptcap.txt)

Ylianst, 2025-02-21:
@mkturner89 So yes, the radio is sending the beacons, not HTCommander. When the beacon is sent, we see that the radio is sending something, but the packet never transits thru the app. So, it does not show on the packet capture. The beacon is sent by the radio regardless if the app is connected or not. So, this behavior is fully expected.

I could turn off the beacon on the radio and have the app send the beacon? Let me know if this makes sense and if you want a different behavior. Also, thanks for sending the logs, that is super useful.

mkturner89, 2025-02-21:
Yeah. you would want the app sending the beacon, not the radio. At least
when connected to it.

Ylianst, 2025-02-21:
Well, I can turn off the beacon on the radio... Is there a usage for this? The problem is, if you disconnect the radio, the beacon will not come back on the radio automatically. Unless we need to customize the beacon in a way that can't be done on the radio, I am not sure why we would not have the radio do the beacon? Thanks.

mkturner89, 2025-02-21:
I'm comparing this to something like the APRSdroid app on my phone. When it
connects, it overrides the APRS settings on the radio, and beacons out
according to the app settings. When I disconnect the app from the radio,
the radio settings go back to what they were.

So the functionality is certainly there, and very useful.
When I'm mobile, I beacon more often as my location changes more
frequently. When connected to the computer, I'm stationary at home, and
only need to beacon every 30 minutes or so to show I'm still active.

Ylianst, 2025-02-22:
Does APRSdroid really disable the radio beacon? Or are both the app and the radio sending beacons?

I ask because APRSdroid uses the KISS protocol which is much more limited. If by chance, you can set the radio to beacon every minute and connect to the AprsDroid and see if this behavior changes, that would be interesting. It would be nice to know if the radio firmware really disables APRS beacon when KISS is connected. Thanks.

mkturner89, 2025-02-22:
I'll check this out and get back to you tomorrow.

khusmann, 2025-02-22:
> Does APRSdroid really disable the radio beacon? Or are both the app and the radio sending beacons?

The latter would be my guess. I haven't played with it much though -- tagging @na7q, he's done much more extensive testing in this area.

Ylianst, 2025-02-23:
@khusmann , @na7q - I have not tested anything, but it looks like this screen has a control on this. I am guessing this switch would cause the radio to stop the beacon and the app would have to do it instead. If true, I need to find this switch and once set, I will beacon from the app.

[image]

mkturner89, 2025-02-23:
I have Synchronous Signaling Settings enabled in the HT app on my phone.
I disconnect the radio from my phone.
 
Radio is programmed to beacon every minute (60 seconds).
It proceeds to beacon once per minute

I launch the Commander App, connect, then touch nothing else.
I am receiving packets, and the radio is still beaconing every minute.

I then set the Beacon Settings in the app to every 30 seconds.
The radio continues to beacon only once per minute.

I then set the Beacon Setting to once every 2 minutes.
The radio starts beaconing only once every 2 minutes.

I then set the Beacon Setting to every 50 seconds.
The radio continues beaconing every 2 minutes.

I set the Beacon Setting back to every 2 minutes.
I then disconnect the app from the radio.

The beacon setting on the radio is now set to every 2 minutes, matching what it was last set to in the app.

mkturner89, 2025-02-23:
I think that all matches with what you were already looking at.
The beaconing settings in Commander are over-writing the radio settings permanently, instead of temporarily.

I dug deeper into the APRSdroid app behavior, and it seems those do their own beaconing, INSTEAD of editing the radio settings.
When I have the radio set to beacon every 2 minutes, and the APRSdroid app set to beacon every 60 seconds, I see BOTH sets of beacons occurring. 

The official HT app seems to have that special setting to allow it to temporarily edit the settings, and put them back after disconnecting. But I'm not sure we would have access to that.

Ylianst, 2025-02-23:
@mkturner89 Ok, this is outstanding data. @khusmann Let me know if you would know of a setting that disabled the radio beacon when the app is connected. Without this, I am not surprised that both APRSDroid and the Radio and sending beacons.

khusmann, 2025-02-23:
> it seems those do their own beaconing INSTEAD of editing the radio settings.

Yes, this is the expected behavior, because APRSDroid works via KISS, so cannot edit the radio settings (with only a couple exceptions, e.g. change channels)

> The official HT app seems to have that special setting to allow it to temporarily edit the settings, and put them back after disconnecting. But I'm not sure we would have access to that.
> @khusmann Let me know if you would know of a setting that disabled the radio beacon when the app is connected.

If the HT app can do it, we can do it. But I'm still not convinced the HT app does this. From my tests, it looks like the "Synchronous Signaling Settings" simply will overwrite the radio's signaling settings with the user's settings on the phone, and does not set them back. For example:

Initial state for both tests:
- Phone is set to beacon at 10min interval (Settings -> ID Signaling)
- Radio is set to beacon at 1min interval

Test 1: Synchronous Signaling Settings ON
- Connect radio to phone -> Radio's beaconing inverval gets set to 10min
- Disconnect radio from phone -> Radio's beaconing interval is still set to 10min
- Final state: Radio settings were overwritten by phone's settings

Test 2: Synchronous Signaling Settings OFF
- Connect radio to phone -> Radio's beaconing setting remains at 1min
- Disconnect radio from phone -> Radio's beaconing setting remains at 1min
- Final state: Radio settings were untouched by phone's settings

@mkturner89 are you seeing something different in your tests?

mkturner89, 2025-02-24:
No. I was able to confirm your findings. I tried changing the APRS settings
from multiple different sub menus, one at a time, and all of them overwrote
the radio settings, even after disconnecting.
The toggle may be broken, or they may be referring to other settings. I
only tried changing the beaconing timer each time.

I think for an app, the important part is that it can handle the beaconing
itself, without touching the radio settings.

khusmann, 2025-02-24:
> The toggle may be broken, or they may be referring to other settings. I only tried changing the beaconing timer each time.

The HT app is not broken; it's just put together in a very confusing way. There are two locations where beaconing settings are made -- the beaconing settings in Settings->ID Signaling will only take affect when Sync is enabled, and overwrite your radio to match your phone. By contrast, "Device Settings->Signaling settings" becomes available when Sync is disabled, and is equivalent to changing the setting directly on the radio (i.e. the radio will overwrite the phone settings here when it first connects). It's not intuitive, and hard to explain.

> I think for an app, the important part is that it can handle the beaconing itself, without touching the radio settings.

Unfortunately, in order to do this, we need a way to get GPS info from the radio, which is currently not possible [see this issue](https://github.com/khusmann/benlink/issues/8). (APRSDroid uses the GPS info from the phone). I've sent an email to BTech about this last week but have not heard back yet. When I do I will update the linked issue above.

@Ylianst  With HTCommander, you could perhaps give users the option to beacon via app or radio? When beaconing via app users could manually set a GPS position or use a 3rd party gps dongle... (But might as well to wait to see what we hear back from BTech)

khusmann, 2025-02-24:
Here's a concise summary of the rules:
- When Sync is enabled and the radio connects with different beaconing settings, the phone setting will overwrite the radio setting
- When Sync is disabled and the radio connects with different beaconing settings, the radio setting will overwrite the phone setting
- When the radio is already connected and a setting on the phone is changed, it overwrites the radio setting, no matter what Sync is set to

mkturner89, 2025-02-24:
I agree that a beaconing option would be nice, with a simple manual GPS
setting for now.

Ylianst, 2025-02-24:
Ok, I will take a look at it. Right now my top priority is to get WinLink support added which is, I hope, fairly important.

@khusmann - Wow, once again, thank you so much for all your information on this. One thing I would like to do with GPS position from the radio: I would pull known WinLink gateways and cache them, then use GPS coordinated to list the nearest ones and attempts to each the gateways automatically.

khusmann, 2025-02-24:
> @khusmann - Wow, once again, thank you so much for all your information on this. One thing I would like to do with GPS position from the radio: I would pull known WinLink gateways and cache them, then use GPS coordinated to list the nearest ones and attempts to each the gateways automatically.

You bet! Gosh that'd be so cool. I sent BTech another ping on GPS the issue today because I haven't heard back yet, but I expect they're taking a while to respond because they're catching up on messages after CNY... stay tuned...

Ylianst, 2025-05-21:
Looks like there is news on this on the [Benlink side](https://github.com/khusmann/benlink/issues/8). If GPS is added to the Bluetooth command set, I will most certainly add support for APRS messaging as part of HT Commander.

Kb1wod, 2025-11-08:
Ylianst, I am using HTCommander with a VR-7500. This thread has helped me get the APRS Beacon to work. Thank you! 
As far as winlink, I have been able to send emails from my radio but not to receive emails. This maybe me not understanding how winlink works. I will keep working on this.

Ylianst, 2026-02-22:
Closing this since it's old, but v0.50 of HT Commander was released. The beacon settings was improved to now configure the channel on which the beacon is sent.

closed 2026-02-22 by Ylianst