Ylianst/HTCommanderLegacy #6: Tone issues when using CHIRP csv

TheCommsChannel, 2025-02-14:
Ylian,

Passing this along from one of the YouTube comments:

> Tried using HT commander with chirp, but when I am importing saved memories, on simplex it adds ctcss tones, and on my saved repeaters it disabled saved tones.  It also only seems to work for a few memory saves, then have to disconnect and reconnect to get more saved.
> 
> AWESOME that someone is working on saving us time who use chirp, and greatly appreciate the work, but passing on info.

I'm not a CHIRP user, but I can create an example csv with different types of entries if needed.

Thanks again for all of your work on this!

TheBravo, 2025-02-16:
That was me who posted that comment that was passed on (very new to github)

If I can be of any help, please let me know. VERY VERY GRATEFUL that you are helping make UB Pro able to import chirp files.

So far what I have tracked.

Power levels imported as low on all, when saved as high
Simplex got ctcss tones added 
Duplex had ctxss tones disabled

Here is a sample of what I am using

[Baofeng_UV-5R_Github.csv](https://github.com/user-attachments/files/18815807/Baofeng_UV-5R_Github.csv)

Ylianst, 2025-02-16:
@TheBravo Thank you SO MUCH for providing a sample file. This removes so much of the guess work. Let me know on that, it will be fixed for the next version.

TheBravo, 2025-02-16:
Sorry, I should’ve thought about that. When I get a chance either later tonight or tomorrow, I will update the file and list the different power settings for like 1, 2 and 3, for low medium high. Currently all those were set to high power. It kind of makes sense to me to maybe default everything to high power because most people do, and then it’s easy to manually adjust to low. 

In CHIRP, before it goes to a CSV file it just lists low, medium high, so I’ve only seen numbers in the CSV file.

And one other issue I’ve noticed. When copying memories over, after I do about 2-3, it stops working. Then I have to disconnect and reconnect to save 2-3 more. And repeat.

Ylianst, 2025-02-17:
Thanks @TheBravo. I just released v0.11 with improvements & fixes. Let me know if it works better. You seem to know a bunch on this topic, I am not sure what to do with "PHX550A" and "FLAG650". The are "CROSS" tone and if you know what I should do, let me know.

[image]

For doing 2 to 3 and stopping, it's probably because I am not doing Bluetooth retry correctly when a command fails to reach the radio over Bluetooth. I need to fix that ASAP. For now, if you bring the radio close to the computer (1 to 2 feet) it may fix it.

TheBravo, 2025-02-17:
> Thanks [@TheBravo](https://github.com/TheBravo). I just released v0.11 with improvements & fixes. Let me know if it works better. You seem to know a bunch on this topic, I am not sure what to do [...]

My radio was less than a foot away from my commuter when I was last programming.

Downloading the update now to test, here is a new csv I made, looks like for a standard UV5R the power options are 1 for low and 4 for high (odd not 5 for 5 watt radio)

When I was saying low-med-high that is an option on the pro but not old UV5

Adding new csv with GMRS as low power.

and not sure about the DTCS, those are additional options for UV5R for local GMRS repeaters, instead of common tones/CTCSS

[Baofeng_UV-5R_Github2.csv](https://github.com/user-attachments/files/18829658/Baofeng_UV-5R_Github2.csv)

and again, THANK YOU for working on this and making this happen. You are going to save so many of us hours of manual programming!!! :D

Ylianst, 2025-02-18:
@TheBravo Nice. Thank you for providing a sample file. I just added power import/export support for Chrip files. For export, I use `1.0W`, `3.0W` or `5.0W` depending on the low/med/high power level. For import, about the same but 1 or less is low, 5 or more is high, rest is medium.

This will be in the next version, don't hesitate to send more instruction on how I could make this better. I will also work on the reliability when I can, super important that it works well.

codywsmb02, 2025-04-15:
Hello!! 

I appreciate your program, and find it super useful! However, for some reason I still cannot import these .CSV files. I can get the channels.csv to open and it will bring up the dialog box to add them. However, the PinellasCOFrequencises.csv won't open the dialog box.

I know you've been working on these, and now it's working for some, but in the .21 version I still wasn't able to open these one's in particular. I'm wondering if that's something on my end, or if it's something still in progress to resolve.

Thank you!

Cody

[channels.csv](https://github.com/user-attachments/files/19762215/channels.csv)

[PinellasCOFrequencies.csv](https://github.com/user-attachments/files/19762125/PinellasCOFrequencies.csv)

Ylianst, 2025-04-16:
Hi @codywsmb02. I just release HT Commander v0.22 and I added this new file format for [PinellasCOFrequencies.csv](https://github.com/user-attachments/files/19762125/PinellasCOFrequencies.csv). This is not like other formats I have seen before, but added it. Let me know if that works for you. Before importing to the radio, you can double click on each to look at that it looks like. Feedback appreciated.

[image]

Ylianst, 2025-04-27:
I will close this, with version 0.23 and 0.24, lots of improvements where made that should fix this. Please keep sending feedback or re-open anytime.

closed 2025-04-27 by Ylianst