Ylianst/HTCommanderLegacy #8: Windows 11 Bluetooth Devices for UV-PRO

rvantrease, 2025-03-26:
It appears that the device UV-PRO name is the same for both of the devices that need to be added.  Windows 11 shows me the second icon, but then when I add it, it collapses back down to the icon for a headset.

rvantrease, 2025-03-27:
> [@rvantrease](https://github.com/rvantrease) I created [this guide here](https://github.com/Ylianst/HTCommander/blob/main/docs/Paring.md) to help with pairing. Let me know if this helps. Indeed HT Commander does not use the headset device but the other non-standard device that seems to be typically hidden in Windows. So, you need to pair with that second one which can be a bit tricky. Other pages that could help [this page](https://www.elevenforum.com/t/enable-or-disable-advanced-bluetooth-devices-discovery-in-windows-11.14237/) and [this Reddit post ](https://www.reddit.com/r/bluetooth/comments/1glh1xt/bluetooth_devices_discovery_setting_missing/).
> 
> Let me know if this helps or if you find something that can help others.

Sorry, still no joy.  It looked like I got the second Bluetooth device added, but then the program wouldn't connect to the radio, then the second Bluetooth device disapeared.

Ylianst, 2025-03-28:
I think worth mentioning is that the radio can only connect via Bluetooth to one application, so, if you phone phone is connected to the radio, HT Commander can't connect. You will need a cellphone icon on the radio if it's already connected. If you see that icon, HT Commander will not be able to connect.

In the Android app, there is a disconnect button here. Use that, then connect HT Commander.

[image]

If that works, I will fix the HT Commander to notify that this is a problem.

Ylianst, 2025-03-28:
HT Commander v0.18 is out.

rvantrease, 2025-03-28:
Thank you for your responsiveness.  I uninstalled the old version, then installed the new version (v0.18), also disconnected the Android app.  I also verified in the paired devices menu on the UV-PRO that my laptop is all that is listed.

Next I ran this PowerShell command on my laptop to verify the Bluetooth connections:
`Get-PnpDevice | where{$_.class -eq 'Bluetooth' -and $_.Name -eq 'UV-PRO'} | ft name, Status, StatusInfo, Present, ClassGUID, HardwareID`

I got these results:
```
Name   Status StatusInfo Present ClassGUID                              HardwareID
----   ------ ---------- ------- ---------                              ----------
UV-PRO OK                   True {e0cbf06c-cd8b-4647-bb8a-263b43f0f974} {BTHLE\Dev_38d200010b06}
UV-PRO OK                   True {e0cbf06c-cd8b-4647-bb8a-263b43f0f974} {BTHENUM\Dev_38D200010B06}
```
However, when I try to click the connect button in HT Commander, nothing happens.  

I think it might be nice to have an option to see what the application sees for compatible Bluetooth devices as a troubleshooting step, and an error message that describes the problem if I press the connect button and it cannot connect.

Thank you again for your responsiveness!

Ylianst, 2025-03-28:
I forgot to ask, what Radio firmware do you have? Also, good PowerShell command! Thank you for that. I have two radios paired to my computer and I see this below. Looks a lot like what you have.

```
PS> Get-PnpDevice | where{$_.class -eq 'Bluetooth' -and $_.Name -eq 'UV-PRO'} | ft name, Status, StatusInfo, Present, ClassGUID, HardwareID

name   Status StatusInfo Present ClassGUID                              HardwareID
----   ------ ---------- ------- ---------                              ----------
UV-PRO OK                   True {e0cbf06c-cd8b-4647-bb8a-263b43f0f974} {BTHLE\Dev_38d20000edd1}
UV-PRO OK                   True {e0cbf06c-cd8b-4647-bb8a-263b43f0f974} {BTHENUM\Dev_38D20000EDD1}
UV-PRO OK                   True {e0cbf06c-cd8b-4647-bb8a-263b43f0f974} {BTHLE\Dev_38d20000faf9}
UV-PRO OK                   True {e0cbf06c-cd8b-4647-bb8a-263b43f0f974} {BTHENUM\Dev_38D20000FAF9}
```

If you enable and go to the debug tab, you should see the radios it finds:

[image]

When you click connect, it should show the connection process. If that does not show anything, I will add more debugging to this. Thanks for helping out debugging. Getting thru the connection process is really important.

rvantrease, 2025-04-01:
I hope I'm helping, not just being a pain in the butt.  I really appreciate the intention of this applicaiton.

OK, I checked the firmware version on my radio.  It shows 0.8.1-2.  I noticed that windows doesn't think the device is connected, so maybe that is part of the problem.

Here is a screenshot of the app with the query devices, plus the help about.  I don't get any output in the debug window when I click on the connect button.

[image]

Here is the Windows 11 Bluetooth settings screen:

[image]

I can get Windows to show connected briefly when I go into the radio menu:
General Settings|Connection
Scroll down under Paired Devices
Hit Select button on my laptop
Check the Call Audio button

Ylianst, 2025-04-03:
@rvantrease - [HTCommander 0.19 is out](https://github.com/Ylianst/HTCommander) with Bluetooth audio support. So, you can listen to the radio on the computer speakers or headset. I had to completely change how Bluetooth works in 0.19, I now use Bluetooth Classic, not Bluetooth LE. Let me know if that fixes anything or if any behavior changes.

Right now, it's audio listen only, I still need to add transmit.

Ylianst, 2025-04-27:
FYI. Version 0.24 is out with more Bluetooth connection improvements. Hopefully it solves this issue.

Ylianst, 2025-05-19:
Ok, I will close this since I have made a bunch more improvements. Please open a new issue if needed. Thanks.

closed 2025-05-19 by Ylianst