Ylianst/HTCommanderLegacy #9: no Possition is displayed in APRS MAP

914tom, 2025-05-14:
Hallo,
I have installed HTCommander in Win10 an connected it to my VR-N76.
I see the incoming APRS packages but get no pins on the map as described.
HEre are the Packages
[image]

But no Pins..

[image]

can anyone help me out

73
tom

Ylianst, 2025-05-14:
I am out traveling now, but once back this weekend, I will check. The packet capture screen you posted if the most interesting since it's not detecting the incoming packets as APRS and so, not decoding the positions. If you happens to select a few of these U-Frame packets, right click on the and save the capture file and attach it here, I can replay the packets on my side and fix it.

Also, just for my info, have you tried any of the other HT Commander features? Speech-to-Text? Terminal? BBS? Torrent? - Thanks.

914tom, 2025-05-15:
Thank you for your response and have a good Time.
That there is no decoding of packages is also coem into my mind, but the raw data contains the Position it seem, at least to me.

[image]

[image]

[image]

BBS connection over the HT working fine 

[image]

Text to speech also working flawless.

Ylianst, 2025-05-15:
Thanks @914tom. This is great information, I will be fixing APRS coordinate on the map.

Can I ask what is the RHBBS? Where can I get this software? It looks nice.

Also, if there are any other features you think would be interesting to add, let me know.

914tom, 2025-05-15:
of course.
It's a BBS Software which is named LinBPQ. 
RHBBS is the Name of 'my' bbs I try to set up for a while, RHBBS its nothing more then the welcome phrese with ASCII Art.
Because there is not really a walk-through setup guide for configuring the software, I don't realy know what I'm doing here *g*. 
The 'TechPrepper' and 'ModenHam' do some, but not realy 'in deep'. 
The Rest I try to figure out with chatgpt. 
It's running 24/7 on my LINUX Home server and I tinker around with it.
73

Ylianst, 2025-05-17:
@914tom - Ok, you are absolutely correct, APRS was not working at all. It's fixed now, I released Version 0.28. It should auto-update and work. Let me know if you see anything else that is not right.

Thanks for the information on BBS. I see some [instructions here](https://www.cantab.net/users/john.wiseman/Documents/InstallingLINBPQ.htm). I am helping that HT Commander and other software will encourage more people to use BBS like this. One think I was wanted to look at is how these BBS do file transfer so I could add support for it.

Ylianst, 2025-05-19:
Ok, closing this since it seems to be solved.