Ylianst/HTCommanderLegacy #10: Issue with importing from CHIRP again - Mac version coming??

TheBravo, 2025-05-20:
Things were off when trying to do a Baofeng UV5R import the other night, I know this is vague, but all I can give you at the moment, because I can't borrow that PC anymore...

So leads into next question, is a MAC version coming soon? :D

Thank you again for everything!

Ylianst, 2025-05-21:
Oh my gosh. So, like the Linux version, that MAC version is possible but it's a bunch of work. For example, I use a native Windows 64 bit SBC encoder/decoder for Bluetooth, etc. My general view right now is to see how much demand there is for this application before putting more work into it. I do have a Macbook Air M1 and I could give it a try at some point and start with minimal features.

TheBravo, 2025-05-24:
Makes sense

I will see if I can get some screenshots next time I play around with on a PC.
But like before, it was throwing off the CTCSS tones, adding a receive and using a different/default one for tx.
And all power settings were bumped down from high.

Ylianst, 2025-07-12:
@DustyLoon  - I am working on a web based version of HT Commander and should work on all OS's since it's a progressive web app (PWA). It's far from done, but you can see the early version at [https://ylianst.github.io/HTCommanderWeb](https://ylianst.github.io/HTCommanderWeb). This version will likely never have the audio support, but it will be able to do APRS, Terminal, Channel configuration and more. If you access it within a Chrome/Edge browser, you can hit the icon at the top and add it as a desktop icon.