Ylianst/HTCommanderLegacy #11: Tone import issues from CHIRP

DustyLoon, 2025-07-10:
When importing a CSV that was created by CHIRP, the tone information is all messed up. ChIRP exports a tone value for both TX and RX, but if tone field is set to "TSQL" then CHIRP only uses the value in the TX field for both RX/TX, but for some silly reason, they put 88.5 in the tone receive field, which puts the wrong RX tone in the radio and you can't hear anything.

There is also a weird issue where when the tone 162.2 is imported for example, the radio gets it as 162.19 (which obviously doesn't work as a valid tone).

Having a way to control/update the radio from a computer instead of the crappy HT app is awesome!

Ylianst, 2025-07-11:
Can you post your CHIRP file? I will fix this. You can cut & paste the file text here. I will use it for testing. Thanks.

TheBravo, 2025-07-12:
Here is a file I just pulled to help out, I noticed this issue as well but just manually corrected.

This would be a HUGE time saver!

[HT Command.csv](https://github.com/user-attachments/files/21199732/HT.Command.csv)

Hmm. wont let me attach a .img chirp file, so here is a dropbox link to it, the above is a CSV of it.

https://www.dropbox.com/scl/fi/90vhg2yxbf6ikbh7wd8l8/HT-Command.img?rlkey=k51gffa4by6up6bzdr0bk11ax&dl=0

Ylianst, 2025-07-13:
Hi @TheBravo  and @DustyLoon . I don't know a lot about CHIRP format, but I do want to make sure work. So, your input is super important. It look like the .img file is radio specific and so, probably not something I will be supporting.

In the .csv file @TheBravo posted, I load it up and see this:

[image]

I am see `Tone` in the `Tone` column, I should use `rToneFreq` for both RX and TX?

Ylianst, 2025-07-13:
Ok, I asked AI for the info on the CHIRP format and got my answer. Working on a fix now.

Ylianst, 2025-07-13:
Version 0.30 is out. Next time you run HT Commander, it should prompt for auto-update if you have auto-update turned on. Let me know if that fixes it. Thanks.

Ylianst, 2025-07-16:
@DustyLoon - Version 0.31 is out, please let me know any feedback at all.

DustyLoon, 2025-07-18:
The update ran in a jiffy, with no extra steps regarding .NET or anything like that. Imported a CSV from CHIRP and it sure seems like everything is importing properly!

That stuff from CHIRP is a mess, you did a great job making this work!

Ylianst, 2025-07-19:
Thank you @DustyLoon - Please don't hesitate to provide feedback on anything else. Your input has been very valuable.

closed 2025-07-19 by Ylianst