Ylianst/HTCommanderLegacy #14: Suggestions

hamMUSings, 2025-09-07:
Hi,

This software is great. I've been successfully using it with my Vero VR-N7500 and Vero VR-N76.  It fills a want I've had since I got the N7500.

I have a few feature suggestions. I have taken a look at the code and tried to figure out if I could contribute code to do the following but I decided not yet :)

1) PTT Keyboard shortcut -- it would be really handy to be able to assign a keyboard button (like space) to act as a PTT instead of having to use the mouse to click on the button in the audio panel.  It would be even better if the user can select the button to use as PTT.

2) GPS "Home" Setting for non-GPS radios -- With the N76 centering in the map was a snap using the built in GPS but with the N7500 which has no GPS having to scroll and zoom each time I open it to my area is less than ideal.  Having an option to hard code the center starting of the map for non-GPS radios would be really handy.

3) Built in hamlib server -- this one I realize would be a fair amount of work.  Though it would be cool to be able to use HTCommander to connect any hamlib enabled software to it and/or for PTT for sound card modes.  I was looking at playing with Vara FM and using virtual sound cables was easy enough to route the audio but I didn't find an easy way to make a PTT trigger.  I'd imagine this would be something like emulate a certain rig command wise and have a server toggle like the webserver and AGWPE server.    

Thanks for what you've done so far and look forward to seeing where it goes.

Ylianst, 2025-09-10:
Hi @hamMUSings and thanks. I am all for looking at HamLib. Can you get me oriented, what API would I need to implement on the HTCommander/HTStation side? - Thanks.

wx4mwa, 2025-11-23:
Is there plans for a Linux build?

wx4mwa, 2025-11-23:
> @wx4mwa <https://github.com/wx4mwa> - Yes, I will give it another try
> this holiday season. I got rid of some Bluetooth Windows x64 DLL so it
> makes a MacOS/Linux port possible. My top priority now is to add a
> soft-modem in HT Commander to try to improve on the speed and reliability
> of the modem in the radio (even give 9600 baud a try).

Cool thanks I will be watching for it.

Matt Andrews